HUST-OS/xv6-k210 #2: 编译可在此系统上运行的程序

Luchangcheng2333, 2021-04-14:
请教下应该怎么样设置编译参数，才能使自己写的程序能在这个系统上运行呢？

SKTT1Ryze, 2021-04-14:
“让自己写的程序在这个系统上运行”与编译参数无关。  
该项目是一个操作系统内核项目，目的是将 MIT 的教学操作系统内核 xv6 移植到 k210 板子上运行，它可以运行一些简单的用户态程序，这些程序在 xv6-user 目录下可以看到。  
让自己写的程序在这个系统上运行的做法是：先参照 xv6-user 目录下的 .c 文件编写源代码，然后通过 makefile 中提供的命令编译成用户态 elf 文件，再放到文件系统镜像中的 /bin 目录下，这样子内核就会识别您写的程序，并可以通过在 shell 里面输入对应的命令运行您写的程序。  
总的来说，这是一件比较复杂的事情。不妨先体验下 xv6-user 中已经写好并调试好的一些程序，比如 ls。

AtomHeartCoder, 2021-04-15:
您好，添加程序的方式已经更新在readme中了，请查看

Luchangcheng2333, 2021-04-16:
> 您好，添加程序的方式已经更新在readme中了，请查看

好的谢谢！

Luchangcheng2333, 2021-04-16:
请问下为什么程序的main函数必须得使用exit(0)函数来结束，而不能在使用return 0呢？使用return 0会报错

AtomHeartCoder, 2021-04-16:
> 请问下为什么程序的main函数必须得使用exit(0)函数来结束，而不能在使用return 0呢？使用return 0会报错

一般我们在win、linux下写的main函数，实际上并不是程序真正执行的第一个函数。在main函数return后，程序也还未真正退出，而是通过exit调用才能退出。您可以通过objdump命令查看一个可执行程序的反汇编结果，就可以看到这些函数。
我们项目中的用户程序没有这么复杂的支持。编译用户程序时，入口地址直接设置成了main函数，因此需要在main函数中直接调用exit才能正常退出。

closed 2021-04-20 by retrhelo