HUST-OS/xv6-k210 #14: 在k210上启动时提示“not FAT32 volume”，请问如何解决该问题？

R1pTheJacker, 2021-05-03:
[image]
上一步已按照README中make sdcard在sd卡中写入了文件镜像

retrhelo, 2021-05-03:
On Mon, 03 May 2021 08:22:57 -0700
R1pTheJacker ***@***.***> wrote:

> [image]
> 上一步已按照README中make sdcard在sd卡中写入了文件镜像
> 

从图片上来看可能是SD卡上的FAT32文件系统不正确。这或许是由于SD卡上存在多个文件分区(Partition)导致的。
可以尝试如下命令将整张SD卡格式化为FAT32文件系统（Linux系统下）
> mkfs.fat -F32 /dev/sdX
其中，`/dev/sdX`为SD卡所对应设备。此举在于确保SD卡下只存在唯一的FAT32文件分区。
进一步的，可以尝试在`kernel/fat32.c`中的`fat32_init()`函数中通过`printf()`打印一些调试信息以便进定位问题。

R1pTheJacker, 2021-05-04:
> On Mon, 03 May 2021 08:22:57 -0700 R1pTheJacker ***@***.***> wrote: [image] 上一步已按照README中make sdcard在sd卡中写入了文件镜像
> 从图片上来看可能是SD卡上的FAT32文件系统不正确。这或许是由于SD卡上存在多个文件分区(Partition)导致的。 [...]

应该不是SD卡FAT32分区的问题，还是同样的报错，我试试调试kernel/fat.c。

retrhelo, 2021-05-04:
这样的话应该还是分区的问题，能出现`sdb1`这样的标识说明SD卡上还是有分区表的。xv6-k210的FAT32驱动是不支持分区表的，默认FAT32文件系统从SD卡的零号扇区开始。可以查查资料看如何删除SD卡上的分区表，然后再重新将整张SD卡格式化为FAT32文件系统。

R1pTheJacker, 2021-05-04:
> 这样的话应该还是分区的问题，能出现`sdb1`这样的标识说明SD卡上还是有分区表的。xv6-k210的FAT32驱动是不支持分区表的，默认FAT32文件系统从SD卡的零号扇区开始。可以查查资料看如何删除SD卡上的分区表，然后再重新将整张SD卡格式化为FAT32文件系统。

感谢，删除分区表后运行成功。

Luchangcheng2333, 2021-05-06:
> > 这样的话应该还是分区的问题，能出现`sdb1`这样的标识说明SD卡上还是有分区表的。xv6-k210的FAT32驱动是不支持分区表的，默认FAT32文件系统从SD卡的零号扇区开始。可以查查资料看如何删除SD卡上的分区表，然后再重新将整张SD卡格式化为FAT32文件系统。
> 
> 感谢，删除分区表后运行成功。

请问下你具体是怎么操作的呢？我的删除了分区表还是出现这个问题

leo-frank, 2021-05-06:
> > 这样的话应该还是分区的问题，能出现`sdb1`这样的标识说明SD卡上还是有分区表的。xv6-k210的FAT32驱动是不支持分区表的，默认FAT32文件系统从SD卡的零号扇区开始。可以查查资料看如何删除SD卡上的分区表，然后再重新将整张SD卡格式化为FAT32文件系统。
> 
> 感谢，删除分区表后运行成功。

请问，可以告知是如何删除分区表的么？是借助哪个软件或是哪个命令。谢谢

R1pTheJacker, 2021-05-06:
> > > 这样的话应该还是分区的问题，能出现`sdb1`这样的标识说明SD卡上还是有分区表的。xv6-k210的FAT32驱动是不支持分区表的，默认FAT32文件系统从SD卡的零号扇区开始。可以查查资料看如何删除SD卡上的分区表，然后再重新将整张SD卡格式化为FAT32文件系统。
> > 
> > 
> > 感谢，删除分区表后运行成功。
> 
> 请问，可以告知是如何删除分区表的么？是借助哪个软件或是哪个命令。谢谢

右键点击sd卡的Properties --> 点击Open in Disks --> 出现下图的窗口 --> 可以点击窗口中“-”符号来删减选中分区
[image]

closed 2021-05-08 by retrhelo