HUST-OS/xv6-k210 #16: 启动到K210时卡死，但是QEMU没事

Luchangcheng2333, 2021-05-06:
我修改了initcode，把init程序硬编码到了内核里面，为什么在初始化时获取当前目录的ename函数时
[image]
紧接着会卡死在申请加锁这里呢？
[image]
并且在QEMU下面没问题，只在K210下有问题

AtomHeartCoder, 2021-05-06:
由于磁盘缓冲层需要进程获取睡眠锁，因此文件系统的初始化是在initcode第一次将要离开内核时进行的，在kernel/proc.c中的代码如下。
```C
// A fork child's very first scheduling by scheduler()
// will swtch to forkret.
void
forkret(void)
{
  static int first = 1;

  // Still holding p->lock from scheduler.
  release(&myproc()->lock);

  if (first) {
    // File system initialization must be run in the context of a
    // regular process (e.g., because it calls sleep), and thus cannot
    // be run from main().
    first = 0;
    fat32_init();
    myproc()->cwd = ename("/");
  }

  usertrapret();
}
```
在`fat32_init`中才会初始化`ecache.lock`，请确保这个函数在`ename`之前进行。

Luchangcheng2333, 2021-05-06:
> 由于磁盘缓冲层需要进程获取睡眠锁，因此文件系统的初始化是在initcode第一次将要离开内核时进行的，在kernel/proc.c中的代码如下。
>
> ```c
> // A fork child's very first scheduling by scheduler()
> // will swtch to forkret.
> void
> forkret(void)
> {
>   static int [...]

感谢

Luchangcheng2333, 2021-05-06:
> 由于磁盘缓冲层需要进程获取睡眠锁，因此文件系统的初始化是在initcode第一次将要离开内核时进行的，在kernel/proc.c中的代码如下。
>
> ```c
> // A fork child's very first scheduling by scheduler()
> // will swtch to forkret.
> void
> forkret(void)
> {
>   static int [...]

那为什么在QEMU下面能正常运行，而K210不能呢？这两个平台好像都用到了这段代码

AtomHeartCoder, 2021-05-06:
在K210上，内存的初始情况是未知的，而QEMU则可能事先将内存清零了，所以锁的未初始化状态可能不同。
由于在FAT32上并不实际存在根目录的目录项（目录项是存放在父目录中的，而根目录没有父目录），因此`ename("/")`并没有真正访问磁盘，而是直接返回它在内存中的静态数据结构。而根目录的结构体会被`fat32_init()`初始化，所以即使你先拿到根目录的结构体，再进行初始化，或许也能使文件系统继续运行下去。

Luchangcheng2333, 2021-05-07:
> 在K210上，内存的初始情况是未知的，而QEMU则可能事先将内存清零了，所以锁的未初始化状态可能不同。
> 由于在FAT32上并不实际存在根目录的目录项（目录项是存放在父目录中的，而根目录没有父目录），因此`ename("/")`并没有真正访问磁盘，而是直接返回它在内存中的静态数据结构。而根目录的结构体会被`fat32_init()`初始化，所以即使你先拿到根目录的结构体，再进行初始化，或许也能使文件系统继续运行下去。

受教了感谢！

closed 2021-05-19 by retrhelo