HUST-OS/xv6-k210 #19: make run platform=qemu

china-zyy, 2021-05-09:
make run platform=qemu, 出现错误， system shutdown scheduled due to RustSBI panic

SKTT1Ryze, 2021-05-09:
如果可以的话把具体的报错信息发一下，如果是 rustsbi 报错应该会有详细的报错信息

china-zyy, 2021-05-09:
好的，具体信息
$ make run platform=qemu
[rustsbi] RustSBI version 0.1.1
.______       __    __      _______.___________.  _______..______   __
|   _  \     |  |  |  |    /       |           | /       ||   _  \ |  |
|  |_)  |    |  |  |  |   |   (----`---|  |----`|   (----`|  |_)  ||  |
|      /     |  |  |  |    \   \       |  |      \   \    |   _  < |  |
|  |\  \----.|  `--'  |.----)   |      |  |  .----)   |   |  |_)  ||  |
| _| `._____| \______/ |_______/       |__|  |_______/    |______/ |__|

[rustsbi] Platform: QEMU (Version 0.1.0)
[rustsbi] misa: RV64ACDFIMSU
[rustsbi] mideleg: 0x222
[rustsbi] medeleg: 0xb1ab
[rustsbi-dtb] Hart count: cluster0 with 2 cores
[rustsbi] Kernel entry: 0x80200000
[rustsbi-panic] hart 0 panicked at 'invalid instruction, mepc: 0000000080200000, instruction: 0000000000000000', src/main.rs:459:17
[rustsbi-panic] system shutdown scheduled due to RustSBI panic

china-zyy, 2021-05-09:
我觉得应该是RustSBI没有装上

china-zyy, 2021-05-09:
我找了好久，都没有装上，请问该怎么装

china-zyy, 2021-05-09:
> 如果可以的话把具体的报错信息发一下，如果是 rustsbi 报错应该会有详细的报错信息

rustsbi-qemu和rustsbi-qemu.bin该怎么使用呢

SKTT1Ryze, 2021-05-09:
项目自带有 rustsbi 二进制文件，在 bootloader/SBI 目录下，分别是 sbi-qemu 和 sbi-k210。  
在 qemu 上运行本项目，只需要运行以下命令：  
```bash
make run platform=qemu
```
从你的报错信息来看，不是 rustsbi 没有装上，而是内核没有装上，正常情况下 0x80200000 处里面放着操作系统内核的二进制文件，而从报错信息中看这个地址上没有东西。  
不知道您是不是按照 readme 中的顺序来操作的，还是做了什么修改。如果是后者，请问做了什么修改。

china-zyy, 2021-05-09:
> 项目自带有 rustsbi 二进制文件，在 bootloader/SBI 目录下，分别是 sbi-qemu 和 sbi-k210。
> 在 qemu 上运行本项目，只需要运行以下命令：
> 
> ```shell
> make run platform=qemu
> ```
> 
> 从你的报错信息来看，不是 rustsbi 没有装上，而是内核没有装上，正常情况下 0x80200000 处里面放着操作系统内核的二进制文件，而从报错信息中看这个地址上没有东西。
> 不知道您是不是按照 readme 中的顺序来操作的，还是做了什么修改。如果是后者，请问做了什么修改。

我clone下来之后，没有修改代码，也是按照readme的步骤做的

china-zyy, 2021-05-09:
> 我clone下来之后，没有修改代码，也是按照readme的步骤做的

china-zyy, 2021-05-09:
> 我clone下来之后，没有修改代码，也是按照readme的步骤做的

装完riscv-gnu-toolchain，之后我修改makefile中的TOOLPREFIX  := riscv64-unknown-linux-gnu-, 因为/opt/riscv/bin下的命令都是riscv64-unknown-linux-gnu-前缀

SKTT1Ryze, 2021-05-09:
> 装完riscv-gnu-toolchain，之后我修改makefile中的TOOLPREFIX := riscv64-unknown-linux-gnu-, 因为/opt/riscv/bin下的命令都是riscv64-unknown-linux-gnu-前缀

这个没问题，内核成功编译了就好。但从报错信息上看 0x80200000 地址上没东西，这就很奇怪了，按理这里应该放的是操作系统内核的二进制文件

china-zyy, 2021-05-09:
> 这个没问题，内核成功编译了就好。但从报错信息上看 0x80200000 地址上没东西，这就很奇怪了，按理这里应该放的是操作系统内核的二进制文件

rcore中也有人遇到这个问题，有人说qemu要5.0.0版本的才行，我重装了qemu 5.0.0也不行

SKTT1Ryze, 2021-05-09:
> rcore中也有人遇到这个问题，有人说qemu要5.0.0版本的才行，我重装了qemu 5.0.0也不行

我们用的是 5.2.0，试试这个版本吧

AtomHeartCoder, 2021-05-09:
可以先尝试`make clean`再重新编译运行。K210和QEMU上的内核入口地址不同，如果当前的编译结果是K210的版本，那么直接在QEMU上跑也有可能出现这样的情况。
最好在编译时也指定目标平台，例如：
```bash
make build platform=xxx
```
实际上，如果代码没有编译的话，直接输入`make run platform=xxx`也会首先进行编译的。此外，可以在`Makefile`中修改`platform`的默认值，在以QEMU为主的开发过程中可以简化命令。

closed 2021-05-09 by china-zyy