HUST-OS/xv6-k210 #23: make build编译过程中报错

potato-particle, 2021-05-22:
[image]
请问这是什么原因怎么解决

retrhelo, 2021-05-22:
请问是对系统做了什么修改吗，比如说更换了链接脚本？这种链接错误一般可以通过修改链接脚本的方式来解决。

potato-particle, 2021-05-22:
没更换过链接脚本只在makefile里删除了riscv64-linux-gcc编译参数里的-mno-relax 加上后会报错 请问大佬用的什么版本的交叉工具链

retrhelo, 2021-05-22:
我现在使用的编译器是是fedora下通过`dnf`获取的工具链。在其他的linux发行版下获取到的工具链应该也能正常工作（项目里的有一位同学就是在Ubuntu下的工具链）。
[image]

potato-particle, 2021-05-22:
请问Ubuntu怎么解决这个工具链的问题啊 可能是我源的问题可我找不到其他源了

AtomHeartCoder, 2021-05-22:
这是我在Ubuntu下使用的gcc工具链版本，可以参考一下
[image]

fuyunxiansen, 2021-05-23:
请问这个工具链在哪里下载？GITHUB上的工具链没找到这个版本

retrhelo, 2021-05-23:
On Sun, 23 May 2021 03:35:11 -0700
fuyunxiansen ***@***.***> wrote:

> 请问这个工具链在哪里下载？GITHUB上的工具链没找到这个版本
> 

直接用Linux发行版自带的包管理软件获取就可以了。比如Ubuntu自带的`apt`。

potato-particle, 2021-05-23:
谢谢大佬 成功了

closed 2021-05-24 by retrhelo