HUST-OS/xv6-k210 #25: mount SD card

potato-particle, 2021-05-23:
请问我在挂载SD卡的时候 挂载点如何选择 内核写入sd卡的内容是什么 是用户态生成的可执行文件么

retrhelo, 2021-05-24:
On Sun, 23 May 2021 04:34:03 -0700
potato-particle ***@***.***> wrote:

> 请问我在挂载SD卡的时候 挂载点如何选择 内核写入sd卡的内容是什么
> 是用户态生成的可执行文件么
> 

目前xv6-k210还不支持挂载SD卡设备。系统内核在启动时就假定了SD卡是可用的（会做一些检查
以确保SD卡确实可用，发现SD卡不可用的话内核会Panic）。而为了能够启动内核，需要提前将一些
用户态程序放进SD卡文件系统的根目录。

potato-particle, 2021-05-24:
[image]
[image]
请问这里已经设置了SD卡为FAT32格式为什么还是会报错

retrhelo, 2021-05-24:
> ![2021-05-24
> 15-03-06屏幕截图](https://user-images.githubusercontent.com/65067401/119313132-c3785100-bca5-11eb-88bf-2a6c22beb249.png)
> 请问这里已经设置了SD卡为FAT32格式为什么还是会报错 
> 

可以看一下[这篇issue](https://github.com/HUST-OS/xv6-k210/issues/14)，里面描述的问题应该和你的一样。

retrhelo, 2021-05-24:
> 请问这里已经设置了SD卡为FAT32格式为什么还是会报错

可以看一下Issue #14，应该和你描述的问题是一样的。

potato-particle, 2021-05-24:
> > 请问这里已经设置了SD卡为FAT32格式为什么还是会报错
> 
> 可以看一下Issue #14，应该和你描述的问题是一样的。

请问我是否有可能修改默认扇区到1上

retrhelo, 2021-05-24:
> 请问我是否有可能修改默认扇区到1上

如果要修改FAT32文件镜像的起始扇区的话，理论上是可以的，你可以尝试修改文件系统的相关代码，尤其是
fat32.c中的代码。但说实在的我个人认为这样的修改是危险且不必要的————它可能会带来许多麻烦，
同时也会使得系统对FAT32文件系统的支持看上去不那么“标准”。所以就我个人而言我并不建议这样做，在
可见的未来xv6-k210不会也不应该提供这样的支持。

potato-particle, 2021-05-24:
好的谢谢大佬

closed 2021-05-24 by retrhelo