HUST-OS/xv6-k210 #27: k210启动初始进程问题

cgxtudou, 2021-05-26:
已经将user中elf文件导入SD卡中  启动时会报错panic：init exiting 请问是什么问题 _init文件也导入sd卡中了

retrhelo, 2021-05-26:
> init exiting 请问是什么问题 _init文件也导入sd卡中了  

在之前的修改中，默认是使用`init`作为启动程序的名称，而不是原本使用的`_init`。可以尝试将`_init`改为`init`，将`_sh`改为`sh`来避免这个问题。

retrhelo, 2021-05-26:
系统启动的时候，第一个用户进程`initcode`会尝试通过`exec`系统调用来运行`init`程序，同样的，`init`程序会通过`exec`调用运行`sh`程序。所以为了能让系统正确地运行起来，需要保证`init`和`sh`两个程序在文件系统中是有效的。

potato-particle, 2021-05-26:
谢谢大佬 初始进程加载进去了 顺便问一下我更改init中的进程调用 把sh换成了其他elf文件会报r_scause寄存器错误是什么原因怎么解决
[image]

AtomHeartCoder, 2021-05-26:
报错的程序是`initcode`，估计是`exec`调用不成功。从`sepc`的值来看，已经超出原`initcode`的代码段了（`initcode`只有几条指令）

potato-particle, 2021-05-26:
不是initcode 我讲init文件换成二进制导入到原initcode数组中没改名字 大小为2.7KB没超页表大小 请问我如何更改他获取的数据段大小不让他超出范围

AtomHeartCoder, 2021-05-26:
堆大小需要通过`sbrk`系统调用扩增，栈大小可以直接在`exec.c`中修改相关代码设置，大概是下面这一段（截取自main分支）
```C
......
sz = PGROUNDUP(sz);
uint64 sz1; 
if((sz1 = uvmalloc(pagetable, kpagetable, sz, sz + 2*PGSIZE)) == 0)
    goto bad;
sz = sz1;
uvmclear(pagetable, sz-2*PGSIZE);
sp = sz;
stackbase = sp - PGSIZE;
......
```

AtomHeartCoder, 2021-05-26:
不过如果你是直接修改`initcode`数组的话，应该是需要修改`userinit()`函数（在proc.c中）

potato-particle, 2021-05-27:
谢谢大佬 在qemu上已经成功将init加载到了内核中并成功运行 但是到k210上运行有时候会有scause=5的加载异常报错 经复位可以避免 请问是什么问题 附init程序
[image]

retrhelo, 2021-05-27:
> 但是到k210上运行有时候会有scause=5的加载异常报错 经复位可以避免

RV 1.9版本下scause=5说明这是一个**Load Access Fault**，产生这种问题的原因可能是多种多样的。
xv6-k210在发生这样的异常如果panic了应该会打印一连串的地址。这串地址可以作为`addr2line`命令
的输入，用来查看程序发生panic之前的运行轨迹，进而定位问题所在。

potato-particle, 2021-05-27:
谢大佬 观察了下发现可能是在init里顺位调用elf造成的撞车问题 经加入循环后发生以下错误请问是什么原因
[image]

retrhelo, 2021-05-27:
> [image]

如截图所示，是所执行的代码中包含非法指令。可以先在QEMU下进行测试，OK了以后再上板测试。建议仔细检查
一下源代码的反汇编结果，以及内核中的硬编码是否正确。

同时，建议检查一下`sfence()`中的代码是否正确。K210下的MMU和TLB存在一些未知的问题，直接使用`sfence.vma`
无法有效地更新TLB。可以看一下Issue #16，里面有讨论这个问题。

luojia65, 2021-05-28:
我觉得提问的好方法是分别寻找或者发一个issue，而不是在同一个issue下面追着作者问。我们的成员并非全职工作，平时还有自己的研究和生活，如果还是以“谢大佬，但是我……这个问题怎么办”的提问方式，你过于打扰朋友们的生活了，极有可能达不到解决问题的效果。
@potato-particle

potato-particle, 2021-05-28:
> 我觉得提问的好方法是分别寻找或者发一个issue，而不是在同一个issue下面追着作者问。更多个issue会促进用户们帮助寻找问题。我们的成员并非全职工作，平时还有自己的研究和生活，如果还是以“谢大佬，但是我……这个问题怎么办”的提问方式，你过于打扰朋友们的生活了，极有可能达不到解决问题的效果。
> 如果还追着我们的成员不放，我将关闭你的所有issue。
> @cgxtudou

抱歉 之前解决问题心切 多有打扰各位

closed 2021-05-28 by luojia65