HUST-OS/xv6-k210 #30: K210 无法进行 SD 卡读取

MRNIU, 2021-05-29:
# K210 无法进行 SD 卡读取

- 描述

    参考此 repo 实现的 SD 卡读写程序，在 `sdcard_write_sector()` 函数中出现 `sdcard: invalid response token` 错误。

- 错误信息

  `sdcard: invalid response token`

- 截图

    [image]

- 复现

    clone `	https://gitlab.eduxiji.net/PTRNULL/simplekernel_oscomp_2021.git`，按照 RUN.md 中的描述运行（K210）即可观察到。

- diff

    代码位于 `https://gitlab.eduxiji.net/PTRNULL/simplekernel_oscomp_2021/-/blob/master/src/drv/sdcard/sdcard.c`
    
    1. 修改了数据类型
    
          `uint8` -> `uint8_t`
    
          `uint16` -> `uint16_t`
    
          `uint32` -> `uint32_t`
    
          `uint64` -> `uint64_t`
    
    2. 注释了锁相关代码
    
        注释了 `static struct sleeplock sdcard_lock` 相关定义，以及依赖它的函数。

retrhelo, 2021-05-29:
从截图上来看是能够与SD卡正常通信的。从截图中可以很明显地看到`SDHC/SDXC detected`以及
`sdcard init`字样，说明SD卡是能够正常地工作的，因此可以初步排除是SD卡本身的问题。

> 2. 注释了锁相关代码
> 注释了 `static struct sleeplock sdcard_lock`相关定义，以及依赖它的函数。

从输出的截图来看系统应该是把K210的两个核都跑起来了的？如果是两个核都跑起来的话是否有可能是
发生了与时间有关的错误？因为SD卡只有一个，由两个核共享，所以属于临界资源，应该通过锁进行保护。
而且对于通信的操作（比如读扇区/写扇区）来说，甚至应该保证其原子性。举一个简单的例子，对于一次
写扇区的操作来说，不可能说进程A先将一半的数据写入SPI的端口，然后被中断打断挂起，等过了一段
时间再调度到进程A时再向SPI端口写入剩下的一半数据。这显然是不行的。

我看到代码中注释掉了sdcard.c中所有与锁有关的代码。不知道在SD卡驱动层的上面是否还有能够保证
原子性的代码，如果没有这方面的代码的话，那么问题就即有可能与此有关。

MRNIU, 2021-05-29:
麻烦的地方就在这里，别说多核了，连多线程都没有，要不然也不敢不用锁啊。
目前的情况是在一张空的 fat32 卡上可以读到0，但是没法写，我还在尝试，但是完全没头绪。
硬件方面我换了两块板子与三张卡，问题可以稳定复现。

retrhelo, 2021-05-30:
> 麻烦的地方就在这里，别说多核了，连多线程都没有，要不然也不敢不用锁啊。

可能是数据传输的过程被中断打断了，如我之前说的那个例子一样。在xv6里面拿到自旋锁之后会关闭中断，保证
临界区代码执行的原子性。

> 举一个简单的例子，对于一次写扇区的操作来说，不可能说进程A先将一半的数据写入SPI的端口，然后
> 被中断打断挂起，等过了一段时间再调度到进程A时再向SPI端口写入剩下的一半数据。这显然是不行的。

就像这样,会不会是这方面的原因?

MRNIU, 2021-05-30:
不是哦，在 `intr init` 中只是填充了一下中断表，没有开启中断。
代码在 `https://gitlab.eduxiji.net/PTRNULL/simplekernel_oscomp_2021/-/blob/master/src/arch/riscv64/intr/intr.cpp`

retrhelo, 2021-05-30:
> 不是哦，在 `intr init` 中只是填充了一下中断表，没有开启中断。

这样，那要不检查一下是否设置了DMA通道？从报错信息来看是`sdcard_write_sector`出了问题。而这里
为了方便是使用了DMA传输数据的。如果不打开DMA的话可以修改一下代码，把所有调用`sd_write_data_dma`函数
的地方改为普通的`sd_write_data`，`sdcard_read_sector`里面的`sd_read_data_dma`也改为普通
的`sd_read_data`。总之就是试试不使用DMA。

如果这样也不行的话，那就试试修改代码循环中的`timeout`，把这个值尽可能调大一点。这样说不定就能解决问题。

MRNIU, 2021-05-30:
ohhhhhhhhhhhh我读到了！
换掉 dma 之后解决了！
谢谢大哥！

closed 2021-05-30 by MRNIU