HUST-OS/xv6-k210 #36: ROM regions overlapping when running on qemu

NPU-Franklin, 2022-03-19:
When I tried to run the kernel on qemu, as a result, I got this error message below.
```shell
qemu-system-riscv64: Some ROM regions are overlapping
These ROM regions might have been loaded by direct user request or by default.
They could be BIOS/firmware images, a guest kernel, initrd or some other file loaded into guest memory.
Check whether you intended to load all this guest code, and whether it has been built to load to the correct addresses.

The following two regions overlap (in the memory address space):
  ./bootloader/SBI/sbi-qemu ELF program header segment 0 (addresses 0x0000000080000000 - 0x00000000800193e8)
  fdt (addresses 0x0000000080000000 - 0x0000000080100000)
make: *** [Makefile:152: run] Error 1
```
It seems that there are some errors with the booting phase. Can you help me fix it? Thanks. Below are some of my compliling system informations.
```shell
..............                                     ********@kali 
            ..,;:ccc,.                             ------------- 
          ......''';lxO.                           OS: Kali GNU/Linux Rolling x86_64 
.....''''..........,:ld;                           Host: VMware Virtual Platform None 
           .';;;:::;,,.x,                          Kernel: 5.16.0-kali5-amd64 
      ..'''.            0Xxoc:,.  ...              Uptime: 6 mins 
  ....                ,ONkc;,;cokOdc',.            Packages: 2991 (dpkg) 
 .                   OMo           ':ddo.          Shell: zsh 5.8.1 
                    dMc               :OO;         Resolution: 1718x900 
                    0M.                 .:o.       DE: Xfce 4.16 
                    ;Wd                            WM: Xfwm4 
                     ;XO,                          WM Theme: Kali-Dark 
                       ,d0Odlc;,..                 Theme: Kali-Dark [GTK2/3] 
                           ..',;:cdOOd::,.         Icons: Flat-Remix-Blue-Dark [GTK2/3] 
                                    .:d;.':;.      Terminal: qterminal 
                                       'd,  .'     Terminal Font: FiraCode 10 
                                         ;l   ..   CPU: Intel i7-9750H (8) @ 2.592GHz 
                                          .o       GPU: 00:0f.0 VMware SVGA II Adapter 
                                            c      Memory: 649MiB / 11940MiB 
                                            .'
                                             .                             
                                                                           
```
```shell
qemu-riscv64 version 5.0.0
Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers

QEMU emulator version 6.2.0 (Debian 1:6.2+dfsg-2)
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers

rustc 1.61.0-nightly (1bfe40d11 2022-03-18)

riscv64-unknown-elf-gcc (gca312387ab1) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

SKTT1Ryze, 2022-03-20:
It may be a problem with the qemu version. Could you please update the qemu-riscv64 version to 5.2.0 or higher and try again?

retrhelo, 2022-03-21:
On my machine, QEMU's version is `5.2.0` exactly. And it just works fine. The development group does find this problem with other versions of QEMU. Maybe it's the problem about this `fdt`, but we still have no idea how to solve it.

You may consult QEMU dev team for further details about this.

NPU-Franklin, 2022-03-22:
I have handed over my current work to one of my friends, I will ask him to test it for me. Thanks a lot! ❤️

closed 2022-03-29 by retrhelo