HUST-OS/xv6-k210 #39: make build过程报错，报错是在xv6-user/sh.c的runcmd函数上

linkadjsks, 2023-04-01:
[image]

retrhelo, 2023-04-02:
我这边使用`riscv64-unknown-elf-gcc (GCC) 9.2.0`可以正常进行编译（使用`main`分支上的代码）。`main`分支下的代码经过多次检验是比较稳定的，应该不会有什么大问题。所以我认为比较可能的原因有两个：

1. **出现编译错误可能和你的编译器版本、选项有关**。要解决这个问题，可以尝试按照报错信息中提示的，在编译选项（Makefile中的`CFLAGS`）中加入对应参数关掉Infinite Recursion的报错；
2. 或者，**这有可能是你修改了sh.c中的代码导致的**。这种情况的话，可能得分析相应的代码才能更进一步的debug。

SKTT1Ryze, 2023-04-03:
> [image]

图中好像哥们用的是`riscv64-unknown-linux-gnu-gcc`编译，可以换成`riscv64-unknown-elf-gcc (GCC) 9.2.0`试试。

linkadjsks, 2023-04-03:
我换成riscv64-unknown-elf-gcc (GCC) 9.2.0后，在make build过程中在新的地方报了错
[image]
我自己拉取的main分支的代码也没有进行任何的更改，编译链的版本信息如下
[image]
我感觉可能是编译链的问题，又或者是操作系统的问题，我是在centos7下进行编译的，实在是混乱了

SKTT1Ryze, 2023-04-05:
看报错是找不到`__sync_lock_test_and_set_4`符号，可以检查下这个符号是项目代码里面的，还是系统库的。如果是系统库的，可能是环境的问题，有些lib库没有安装；如果是项目代码里面的，需要问下另外一个维护者 @retrhelo 。

SKTT1Ryze, 2023-04-05:
> 看报错是找不到`__sync_lock_test_and_set_4`符号，可以检查下这个符号是项目代码里面的，还是系统库的。如果是系统库的，可能是环境的问题，有些lib库没有安装；如果是项目代码里面的，需要问下另外一个维护者 @retrhelo 。

但是现在这个仓库的维护者都毕业了，不确定能否抽出时间，抱歉抱歉 😭

linkadjsks, 2023-04-05:
> > 看报错是找不到`__sync_lock_test_and_set_4`符号，可以检查下这个符号是项目代码里面的，还是系统库的。如果是系统库的，可能是环境的问题，有些lib库没有安装；如果是项目代码里面的，需要问下另外一个维护者 @retrhelo 。
> 
> 但是现在这个仓库的维护者都毕业了，不确定能否抽出时间，抱歉抱歉 😭

没事大佬，前面的问题我已经解决了，应该是编译链的问题，我在ubuntu下apt安装`binutils-riscv64-unknown-elf`和`gcc-riscv64-unknown-elf`后成功编译了。

但是现在又遇到了新的问题，我在fat32格式的sd卡里拷贝了用户程序
[image]
[image]
插入开发板后，运行make run的时候卡住了
[image]
这问题是sd卡读不到吗，还是开发板的问题

SKTT1Ryze, 2023-04-05:
这个应该是拷贝方式不对，具体细节我也忘了。可以参考下这里：https://github.com/HUST-OS/xv6-k210#run-on-k210-board

retrhelo, 2023-04-06:
这类问题可以通过内核反馈的backtrace来定位问题所在（就是图中`backtrace`下面的一系列地址）。这些地址标记了发生panic时内核中的执行路径。

此外，在真正进行上板开发之前，可以考虑先在QEMU上进行仿真验证。

linkadjsks, 2023-04-11:
运行make run时卡住原来是因为大小写的问题，问题出在fat32.c中的dirlookup()上，不知道为什么我的sd卡里小写的文件名读出来全是大写，导致找不到init程序，所以初始化的时候就进行不下去了。
我在qemu上运行时，dirlookup()里的情况是正常的
[image]
在K210上运行时，读的情况很奇怪，而且大小写不匹配导致找不到对应程序，所以运行不下去
[image]
我在dirlookup()`里强行把大写转成小写后终于成功启动了，尝试几个命令都没有问题。感谢上面两位大佬的细心指导！

closed 2023-04-11 by linkadjsks