HUST-OS/xv6-k210 #42: k210 是否实现了缓存一致性？

BigBrotherJu, 2024-03-04:
如果没有实现缓存一致性，多核运行可能会有问题吧？

retrhelo, 2024-03-04:
对于多核来说是有内存一致性是必须实现的，在xv6-k210中主要通过fence指令来完成。可以参考[spinlock.c](https://github.com/HUST-OS/xv6-k210/blob/9f1c1bc57957ae0a59f4501ae9b6257d2648af11/kernel/spinlock.c#L62)中的代码和注释。

BigBrotherJu, 2024-03-04:
我说的是cache coherence，比如两个核读取同一个全局变量，这个变量被放在两个核的 cache里，然后一个核改变这个变量的值，两个 cache 必须保证不起冲突。

retrhelo, 2024-03-05:
我认为注释里的“to ensure that all the stores in the critical section are visible to other CPUs”已经足够回答你的问题了。

BigBrotherJu, 2024-03-05:
我说的是硬件上 k210 这个 soc 是否实现了 cache coherence，我翻了一下 k210 的手册，好像没有提到实现了。

SKTT1Ryze, 2024-03-05:
> 我说的是硬件上 k210 这个 soc 是否实现了 cache coherence，我翻了一下 k210 的手册，好像没有提到实现了。

这是一个开源的操作系统内核项目，处理器/硬件层面的缓存一致性一般来说会交给硬件厂商处理，实现细节的话朋友可以咨询下 k210 的制作商。感谢反馈。: )

BigBrotherJu, 2024-03-05:
那如果硬件没有缓存一致性，多核运行 xv6 会有问题吧，你们运行下来有很奇怪的 bug 吗？如果没有的话，可能厂商实现了缓存一致性，没有在手册中提到。:)

SKTT1Ryze, 2024-03-05:
> 那如果硬件没有缓存一致性，多核运行 xv6 会有问题吧，你们运行下来有很奇怪的 bug 吗？如果没有的话，可能厂商实现了缓存一致性，没有在手册中提到。:)

目前为止我们没有遇到过因为缓存一致性导致的bug，但也不能保证在后面的使用中不会遇到，对于任何软件来说将来遇到的bug都是未知的。朋友如果后面遇到问题，也欢迎在此repo下发issue，我们会尽量抽出时间一起来解决问题。感谢朋友的理解和支持。: )

closed 2024-03-05 by BigBrotherJu