HUST-OS/xv6-k210 #43: 如何跑通测试样例文件夹

BeiNing-Z, 2024-03-25:
我现在已经有了测试样例的文件夹riscv64，该怎么在qemu上跑通测试呢，该将他放在哪里，且用什么指令行呢

retrhelo, 2024-03-26:
你需要制作一个fs.img镜像，可以参考一下Makefile里`fs`的相关部分。

BeiNing-Z, 2024-03-26:
> 你需要制作一个fs.img镜像，可以参考一下Makefile里`fs`的相关部分。

我已经制作了一个fs.img,是要将测试文件夹放入里面bin/下面嘛？

retrhelo, 2024-03-26:
理论上放在哪里都行，只要你的内核在启动的时候能找到这些程序（这可能会需要修改initcode）。

BeiNing-Z, 2024-03-26:
> 理论上放在哪里都行，只要你的内核在启动的时候能找到这些程序（这可能会需要修改initcode）。

我在运行测试的时候会出现以下这种错误-> /bin/riscv64 $ exit
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 220
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 93
pid 5 exit: unknown sys call 260
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 64
pid 5 exit: unknown sys call 93
[rustsbi-panic] hart 1 panicked at 'invalid instruction, mepc: 0000000000000456, instruction: 0000000000000000', src/main.rs:459:17
[rustsbi-panic] system shutdown scheduled due to RustSBI panic

retrhelo, 2024-03-26:
这看上去像是对应的syscall没有实现导致的。和应用程序在文件系统的位置无关。

BeiNing-Z, 2024-03-27:
> 这看上去像是对应的syscall没有实现导致的。和应用程序在文件系统的位置无关。

  能详细说一下具体如何跑通测试吗？我已经存在了一个fs.img,然后成功运行了make run platform=qemu。

retrhelo, 2024-03-27:
很简单，可以通过以下三个步骤跑通测试：

1. 阅读本项目代码和比赛要求；
2. 根据比赛要求编写代码，自己实现测试用例中需要的syscall；
3. 跑通测试。

**在比赛中遇到的未实现的syscall需要你自己去实现，xv6-k210的维护者没有义务替你去做这些工作。**

closed 2024-03-27 by retrhelo