HUST-OS/xv6-k210 #44: QEMU初始化运行make run pltaform=qemu时会panic

BeiNing-Z, 2024-04-02:
qemu-system-riscv64 -machine virt -kernel target/kernel -m 6M -nographic -smp 2 -bios ./sbi/sbi-qemu -drive file=fs.img,if=none,format=raw,id=x0  -device virtio-blk-device,drive=x0,bus=virtio-mmio-bus.0
ekernel: 0x8000f000
heap_start: 0x8001a000
heap_size: 0x4000
trap init

________   ________   ___   ________   ________   ________   ________   ___     
|\   __  \ |\   ____\ |\  \ |\   ____\ |\   __  \ |\   ____\ |\   __  \ |\  \    
\ \  \|\  \\ \  \___|_\ \  \\ \  \___| \ \  \|\  \\ \  \___|_\ \  \|\ /_\ \  \   
 \ \   ____\\ \_____  \\ \  \\ \  \     \ \   __  \\ \_____  \\ \   __  \\ \  \  
  \ \  \___| \|____|\  \\ \  \\ \  \____ \ \  \ \  \\|____|\  \\ \  \|\  \\ \  \ 
   \ \__\      ____\_\  \\ \__\\ \_______\\ \__\ \__\ ____\_\  \\ \_______\\ \__\
    \|__|     |\_________\\|__| \|_______| \|__|\|__||\_________\\|_______| \|__|
              \|_________|                           \|_________|

[PsicaSBI]: Version 0.4
mideleg: 0x222, medeleg: 0xb1ab
Extension: RV64ACDFIMSU
PsicaSBI hartid 0 ready
  (`-.            (`-.                            .-')       ('-.    _   .-')
 ( OO ).        _(OO  )_                        .(  OO)    _(  OO)  ( '.( OO )_ 
(_/.  \_)-. ,--(_/   ,. \  ,--.                (_)---\_)  (,------.  ,--.   ,--.) ,--. ,--.  
 \  `.'  /  \   \   /(__/ /  .'       .-')     '  .-.  '   |  .---'  |   `.'   |  |  | |  |   
  \     /\   \   \ /   / .  / -.    _(  OO)   ,|  | |  |   |  |      |         |  |  | | .-')
   \   \ |    \   '   /, | .-.  '  (,------. (_|  | |  |  (|  '--.   |  |'.'|  |  |  |_|( OO )
  .'    \_)    \     /__)' \  |  |  '------'   |  | |  |   |  .--'   |  |   |  |  |  | | `-' /
 /  .'.  \      \   /    \  `'  /              '  '-'  '-. |  `---.  |  |   |  | ('  '-'(_.-'
'--'   '--'      `-'      `----'                `-----'--' `------'  `--'   `--'   `-----'
hart 0 init done
PsicaSBI hartid 1 ready
hart 1 init done

scause 0x000000000000000d
sepc=0x000000008002560c stval=0x0000000000000020 hart=0
pid: 2, name: initcode
[trap]: hart 0 at kernel/trap/trap.c: 236
[panic]: kerneltrap
backtrace:
0x000000008002033e
0x000000008002fbd8
0x000000008002f7a0
0x000000008002bba0
0x000000008002ec22
0x000000008002cefa

SKTT1Ryze, 2024-04-02:
看输出好像您报错的项目不是本项目，而且从输出里的“ PsicaSBI”看这也不是本项目会出现的报错。所以这里给您两个建议：
1. 在https://github.com/oscomp/2021oscomp-best-kernel-design-impl/tree/2021-xv6 项目下开启issue询问下项目作者；
2. 看下kernel/trap/trap.c的236行代码，尝试分析下可能的引起panic的原因

没能提供解决方案非常抱歉（我已经很久没有关注这个项目了），感谢反馈🙏

retrhelo, 2024-04-02:
PsicaSBI是xv6-k210用到的SBI实现，但是从报错信息上来看，此处内核已经启动成功了，应该不是SBI的问题。

retrhelo, 2024-04-02:
> 看输出好像您报错的项目不是本项目，而且从输出里的“ PsicaSBI”看这也不是本项目会出现的报错。所以这里给您两个建议：
> 
>     1. 在https://github.com/oscomp/2021oscomp-best-kernel-design-impl/tree/2021-xv6 项目下开启issue询问下项目作者；
> 
>     2. 看下kernel/trap/trap.c的236行代码，尝试分析下可能的引起panic的原因

翻了一下代码，根本没找到kernel/trap/trap.c这个文件。我也建议你按照上面这两个建议去做。

closed 2024-04-03 by retrhelo